im-tomu/tomu-hardware #7: Test the ARM SWD

mithro, 2016-09-16:
Test that the CPU can be debugged via the ARM SWD pins using OpenOCD and GDB.

maks, 2017-01-20:
@mithro which pads on the board are the SWD pins brought out to?
I want to have a go at using SWD with a buspirate and openocd to do the initial flashing [per this guide](https://www.earth.li/~noodles/blog/2015/08/program-fst01-with-buspirate.html).

sa-wilson, 2017-01-27:
@maks C and D are the SWD pins. CLK and DIO respectively.

I screwed up the UART bootloader upgrade and then had to use OpenOCD and an STLink-v2-1 to flash the bootloader back, so I know that particular combination works. Only OpenOCD 0.10 has Happy Gecko support though. I didn't try any debugging.

mithro, 2017-02-03:
Instructions on using ARM 2-wire with OpenOCD on a RPi are now at
https://github.com/im-tomu/tomu-bootloader/tree/master/openocd

It's definitely working, so closing this issue.

closed 2017-02-03 by mithro