im-tomu/tomu-hardware #8: Create a easy programmer rig

mithro, 2016-09-16:
Something with spring-loaded contacts / pogo pins....

mithro, 2016-09-16:
Needs a USB->Serial interface to enable loading of the initial firmware.

maks, 2017-01-20:
Only just started reading the [mcu's documentation](https://www.silabs.com/Support%20Documents/TechnicalDocs/EFM32HG309.pdf) and initially spotted: 

> 2.1.13 Pre-Programmed USB/UART Bootloader

Is there a reason why this can not be used for easy firmware flashing?

maks, 2017-01-20:
Reading the [app note](https://www.silabs.com/Support%20Documents/TechnicalDocs/AN0042.pdf) you can use either USB or UART, so I guess next question is are the T/R pins USART0 ?

joshgoes, 2017-01-20:
> Is there a reason why this can not be used for easy firmware flashing?

Yes - the pre-programmed USB/UART Bootloader depends on an external high-frequency XTAL for USB functionality, despite there being a perfectly good internal high-frequency XTAL for USB. This is an oversight on Silicon Labs' part. As the Tomu has no external XTAL, the USB functionality is nonexistent. This is why we have to use the (Silicon Labs provided) AN0042 bootloader. It is more-or-less identical to the factory-programmed bootloader but with the XTAL stuff fixed up, and the version string changed slightly.

> Reading the app note you can use either USB or UART, so I guess next question is are the T/R pins USART0 ?

The UART mentioned in the bootloader docs is actually an LEUART, which is wired to the D/C pins. This means that the D/C pins are used by both SWD as well as a low-energy UART.  The T/R pins are wired to a different UART to the D/C pins. This allows people to have SWD and UART (or two UARTs) going at the same time.
I can't remember in which doc (or where in the doc), but there's a table of all the peripherals plus their "locations" -- with pins. Check out the Tomu schematic to find which pins T/R are wired to, and then check the table to see which UART+Location combos are available at those pins.

maks, 2017-01-20:
Now that @mithro emailed out about the project discuss google group, I see the post there from @joshgoes already covers how to use the builtin bootloader to flash via uart.

kevinb456, 2018-11-14:
I've just noticed this on a browse of interesting kitspace.org projects - I have had success in my work capacity in designing simple programming sprung-probe rigs so could share some ideas for this.

I've found that a mechanical base designed in Freecad, and printed on a 3D printer can be quite effective. Probes/sleeves such as Harwin P19-4021 and S19-501 can contact to the SWD pins, and can be inserted straight into the 3D printed base if designed with a tight fit. Also attach a clamp such as Brauer H100/2 to the base which brings a second 3D printed 'press' part down on top of the PCBA. Then combining with a programmer such as an LPC-Link2, and the whole thing bundled up inside a plastic enclosure works well - insert PCBA, close press with handle, program PCBA, open press & remove... rinse&repeat! Depending on volume this could be expanded to do a panel at once with multiple programmers.