im-tomu/tomu-hardware #9: Figure out a way to add a good random value source

mithro, 2016-09-16:
Something like the http://altusmetrum.org/ChaosKey/

FlorianUekermann, 2018-02-06:
Maybe there are ways to do this without additional hardware. Would something similar to the approach Gniibe takes with [NeuG](https://www.gniibe.org/memo/development/gnuk/rng/neug.html) be an option?

Documenting an approach that is straightforward to implement would be great to get some peer review and avoid broken homemade rngs.

mithro, 2018-02-06:
Keith Packard was suggesting that we use a good random number source to bake some random data into the flash. Then as you use the random data you can clear it until you run out. As you only need the true random data to seed a pseudo random number generator it should last quite a while...

mithro, 2018-02-06:
I'm no security / crypto expert however

FlorianUekermann, 2018-02-06:
That sounds fine and fast for cases where you don't need cryptographically secure numbers. However, the application I had in mind was crypto.

So some (slow) source of true randomness would be great to seed a CPRNG with low frequency. Not an expert either, so I would very much like to hear what Keith, Gniibe or other experts think.

gl-sergei, 2018-02-07:
There is Neug port for Tomu here: https://github.com/im-tomu/chopstx/blob/efm32/mcu/adc-efm32.c and https://github.com/im-tomu/chopstx/blob/efm32/u2f/neug.c), would be great if someone could audit it.

There is also Fortuna PRNG implemented by Si Labs https://www.silabs.com/documents/public/application-notes/AN0806.pdf for your consideration.