im-tomu/tomu-hardware #12: Make the tomu PCB work with OSH park

mithro, 2016-11-24:
See http://docs.oshpark.com/submitting-orders/cutouts-and-slots/

Currently the slot doesn't work;

[image]

flummer, 2016-12-30:
I'll be happy to look into this... Since this slot isn't plated (that would short two of the pads), this can be milled along with the outline, so if the hole is replaced with an additional outline (inside cut-out), the same file should work with OSH park and the other fab houses.

I'll be happy to do that and send a pull request, just need to separate some of the stuff in the current design first.

pdp7, 2017-02-14:
Hi @flummer, thanks for making this change.  I work for OSH Park and have just ordered your modified version: https://oshpark.com/projects/4tCH773A

I used our 0.8mm thickness service.  Our normal 1.6mm thick 2 layer panels go out daily, but this one is less frequent.  The current 0.8mm panel is already at 50%, so it should be going to fab on Friday.  I should have it be the end of February. 

I'll order an OSH Stencil and the parts and attempt to assemble.  I've used paste before with SOIC but not a QFN, so it will be an interesting experience.

bokowski, 2018-02-26:
One year later... I thought I'd update this issue saying that the board (OSH Park link above) works.

vincenttl, 2018-03-27:
Hi, I build a Tomu from oshpark: https://oshpark.com/shared_projects/ZAMlmpMB 
It seems to be working. 
Got the bootloader running and this too https://github.com/im-tomu/chopstx/tree/efm32/example-tomu

The touch seems to work fine. Anything else I should try?

mithro, 2018-04-02:
Got some pictures? Did the hole/slot work?

vincenttl, 2018-04-02:
Here is one of the 3. It would need to be cleaned up a little bit.
I also left it a bit too long on the hot plate for soldering.

I continued playing with it and there might be an issue.
I flashed the dfu bootloader without issue but it comes back to the
bootloader every power cycle even if I used dfu-utils to program the u2f
firmware. I checked for a short between the outer pads and they are not
shorted.

I'll investigate soon.

todbot, 2018-04-02:
The bootloader always running is not a short in your circuit.  It's apparently by design: https://github.com/im-tomu/tomu-bootloader/issues/31