im-tomu/tomu-hardware #13: Order and test tomu boards from Dirty PCBs

mithro, 2016-11-24:
http://dirtypcbs.com/

mithro, 2016-11-24:
Your order number:2027981
```
PCBs	File	Layers	Size (actual)		Color	Thickness	Coating	Stencil	Processing	Quantity	Charge	Status
28514	xx	2	5x5cm(1.1x1.3cm)	Yellow	1.0mm		HASL	None	Normal		Protopack ±10	$14	Order received
	 	 
Shipping	Hongkong Airmail (1-8+ weeks)	$0	0.1kg
Total	 	$14	 
xxx - tomu-v0.2-11-g07748ae.zip
```

mithro, 2016-11-24:
[image]

luftek, 2017-01-28:
DirtyPCB do panelized 2 layer boards. Like:
[image]

I use https://github.com/ThisIsNotRocketScience/GerberTools , works great. Their tools are also used for rendering the DirtyPCBs

A append an example of your v0.2 board but multiplied on a 10x10cm prototype panel. Roughly 36 times 10-12 panels which is around **400pcs for 25$ (panel + shipping)**
Your panelized board would look like that:
- choose red color :)
- breaktabs can be made smaller, for easier depanelization
[image]

I make a pull request?

mithro, 2017-01-30:
Make sure you order the thinner PCB - please do send a pull request.

luftek, 2017-01-30:
I don't have a use for im-tomu right now -> won't order samples.
Just wanted to let you know. Tell which size you choose for panelizing (5x5 OR 10x10 cm) and I will do it (pull request).

mithro, 2017-01-30:
Lets do both?

luftek, 2017-01-30:
Sure