im-tomu/tomu-hardware #29: Schematics image, but with a photo/diagram of the physical layout

micolous, 2018-03-08:
I was having a look to see what is brought out from the EFM32HG309 on the back of the board, and so my first port of call was `/docs/tomu.pdf`.  But this shows the logical layout.  There's also no indication about which version of the board it applies to.

What would help is if there were pictures/diagrams of the actual, physical layout of the board, pointing to what each pad corresponds to on the EFM32HG309.

This would help people who are less skilled with electronics to know what they're wiring on to.

I _think_ what it is:

- The 4 pads on the top labelled with "D" and "C" are the debug interface, and, from left to right:
  1. 3.3v
  2. SWDIO
  3. SWDCLK
  4. Ground (square)
- The 4 pads on the bottom, labelled with "T" and "R" are the UART interface, and, from left to right:
  1. 3.3v
  2. PB13 / LEU0_TX
  3. PB14 / LEU0_RX
  4. Ground (square)
- The four contacts on the left edge are the touch-sensitive pins, not sure what is connected to what there, but could also be repurposed for analogue or digital I/O.

mithro, 2018-03-08:
There is some info here -> https://github.com/im-tomu/tomu-bootloader/tree/master/openocd#wiring

mithro, 2018-03-08:
We would love for someone to create pin diagrams like those found on http://fritzing.org/parts/ or http://diwo.bq.com/pinout-uno/

PR #30 (2018-03-18): Documentation improvements
- Adds 3D renders of the board, and instructions on how to make new ones.

- Adds a pinout diagram (SVG and PDF versions included), and instructions on how to make new ones.

  The pinout diagram includes notes on what's changed since v0.3 (LCA 2018 build).

- Normalises some Markdown formatting.

- Moves all PCB design files to `./pcb`.

- Updates `release.sh` to point at new paths, and points it at kitspace.yaml (from commit 33289ad)

- Adds .DS_Store to .gitignore.

Fixes #29
closed 2018-03-18 by xobs